rt-bishop/Look4Sat #1: Request apk file

ntoulasd, 2020-02-10:
Could you also provide apk file. We dont all have google play accounts.

Thanks

agervail, 2020-02-10:
Yes, I was going to ask exactly the same thing, if you could do a release once in a while it would be greatly appreciated !!!

agervail, 2020-02-10:
You could also upload it on F-droid as it is open source ;) so we can get the updates !

rt-bishop, 2020-02-11:
Hello, everyone!
Have just added an apk file to the release-apk folder in the repository, as you requested.

To release it on F-Droid I still need to get rid of GooglePlay location services dependency. I'll do this (and a lot of fixes) as soon as I can. Currently I have just an hour or so of the free time in the mornings for this app, but hopefully will do another release within a week.

Thank you very much for your interest to this app. All your feedback is hugely appreciated!

closed 2020-02-11 by rt-bishop

agervail, 2020-02-11:
Great, thanks.
Maybe if you have time you can also do a "github release" there : [github release](https://github.com/rt-bishop/Look4Sat/releases)
Your app is great !!