rt-bishop/Look4Sat #2: Use Maidenhead locator as source for location.

petrus-lt, 2020-02-10:
@rt-bishop, this is app looks really great. 

One nice feature I guess would be to get the location from a known maidenhead locator. This would allow the app to run almost without any privileges. 

Once a maidenhead lib is included, you can easily show the grid locator from the location if allowed by the user :)

agervail, 2020-02-10:
Can't find any android libraries doing that. But it looks easy to compute the coordinate with a simple formula.
What about what3words that looks like it's used a lot on the playstore ?

petrus-lt, 2020-02-10:
what3words is a company and using a proprietary algorithm. Nothing prevents them to change the words tomorrow. It might need an active internet connection. And how do I know if entrepôt.formuler.magique (so easy to spell, write in cw or use in ft8..) is near vermeil.entier.riveter? Using Maidenhead I know that N is around my latitude, CONUS is between ~C and E, Europe between I-K etc.
For those reasons (and obviously maidenhead is older), what3words is not really helpful in such an app.

Sure, if you want to have fun implementing it, please go ahead. But its usefullness is doubtful.

agervail, 2020-02-10:
I think you're absolutely right, didn't think of all those points.
I have to agree with you.
I will take a look at the application code to see if I can help but I'm definitely not an android developper....

rt-bishop, 2020-02-12:
Hey, guys!
I will definitely do it, already have some ideas about the ways to add this functionality. However, the priority for now is bug fixes, the reports show quite a few bugs that I can't reproduce on any of my devices. Once they are fixed, I'll proceed to the locator implementation.

joeswhite, 2020-02-12:
> Hey, guys!
> I will definitely do it, already have some ideas about the ways to add this functionality. However, the priority for now is bug fixes, the reports show quite a few bugs that I can't reproduce on any of my devices. Once they are fixed, I'll proceed to the locator implementation.

Are you married to a specific location provider like maidenhead?
Also, and more importantly, you talk about being unable to reproduce bugs you are being informed about, I'm trying to find the references in some form or either an issue on github or a notes file for these bugs so I can replicate them and document the steps taken to replicate them and fix them. Where are the bugs referenced so I can view them and assist since you are limited on time (that's if you want help that is).

rt-bishop, 2020-02-12:
I'm getting quite a few (over 140 tbh) crash reports from the Google Play developer console with stack traces and all that, but most of the crashes are hard to reproduce. That would be epic to get any proper bug reports, but not all the users can or want to do that. Also there are some bugs in predict4java library that lead to certain TLE items cause an indefinite loop calculation. I'm about to fix that, potentially will rewrite the whole library to Kotlin.
I'm very new to all this, as this is my first proper Android and open-source project, so all your help is hugely appreciated! Hopefully the recent update, v1.2.2, will cover most of the problems. I'll keep an eye on the dev console, I'd like to have things more or less stable before trying to implement any additonal functionality.

joeswhite, 2020-02-13:
> I'm getting quite a few (over 140 tbh) crash reports from the Google Play developer console with stack traces and all that, but most of the crashes are hard to reproduce. That would be epic to get [...]

If you are interested, I can open a new issue reguarding all errors that you are being notified of so they can be discussed so this thread doesn't get cluttered? Then we can pick each problem apart if you want to post some of the common problems there? I'm working through older versions of Android on old devices attempting to replicate crashes and would like to document what I've tried so far

rt-bishop, 2020-02-19:
Things seem relatively stable at the moment, so I can start adding features. If you experience any bugs or crashes all the bug reports will be greatly appreciated.

jcp-sd, 2020-09-21:
Everyone likely knows the calculation of lat/lon for a given Maidenhead is almost trivial.

Not a biggie as there are many helper tools out there to do this separate from the program.

Attached is a short Perl script I often use for that purpose to demo the algorithm.

Reference to the original can be found on Wikipedia's page on Maidenhead locator.

73, John

[grid_to_latlon.txt](https://github.com/rt-bishop/Look4Sat/files/5258025/grid_to_latlon.txt)

rt-bishop, 2020-09-21:
Hey John! I'm working on a complete UI rework at the moment, but I'll make sure to include the Maidenhead locator calculation and display with the next update! Thanks for the reminder, trying to find time for everything)

closed 2020-11-29 by rt-bishop