rt-bishop/Look4Sat #3: Publish on F-Droid

der-stefan, 2020-02-12:
Great piece of software! What do you think about publishing it on F-Droid additionally?

rt-bishop, 2020-02-12:
I have plans to do that, but first I need to get rid of Google Play location services dependency and switch to the native api. I think I'll do it within a week or two, struggling with free time at the moment. But yeah, it'll be on F-Droid soon.

der-stefan, 2020-02-17:
Great, thank you!

Symbian9, 2020-03-21:
JFTR, here is its page in *IzzyOnDroid Repository*

- https://apt.izzysoft.de/fdroid/index/apk/com.rtbishop.look4sat

IzzySoft, 2020-04-24:
@rt-bishop could you please fix the changelogs in your Fastlane structure? For the latest release (`versionCode` 11), there's no `11.txt` – and the one that is there doesn't match any existing `versionCode`. My updater just reported it missing :wink:

IzzySoft, 2020-05-27:
@rt-bishop Fastlane changelogs are still broken. Gradle gives versionCode 14, and there's no `14.txt` – instead there are `100000.txt` and `100001.txt` which match no existing versionCode. From the timestamps I assume the latter should match the current release? They are no use if they cannot be matched.

rt-bishop, 2020-05-28:
@IzzySoft Thank you very much for letting me know. Somehow completely missed all the notifications on the github issues. Also didn't realise that the filename should match the version code. It's done now.

IzzySoft, 2020-05-28:
@rt-bishop for your convenience: [My Fastlane Cheat-Sheet](https://gitlab.com/snippets/1895688) :wink:

And wasn't that `15.txt` for VC 14? I manually fetched it just after that release, taking the latest there was IIRC. But as the names couldn't be matched, it's of course hard to tell – though the content of `15.txt` matches what the release comment for VC 14 says. Ah, well, I see: the only change between the two is the commit fixing Fastlane :smile:

I've updated the RFP accordingly and pinged our packager again, giving him some details. Hopefully Likaon-Kter picks it up soon now.

closed 2020-06-18 by rt-bishop