rt-bishop/Look4Sat #4: Request: select all sats

novakmi85, 2020-02-18:
Could you please add "SELECT ALL" feature to select all satellites? Thank you. (BTW: Great software, thank you very much!)

rt-bishop, 2020-02-19:
Hey) Actually working on it at the moment, as wel as on the search function. Will add more satellites after that.

trebor-56, 2020-02-19:
Option to centeralise map 3 options should be ample 
as is, Africa, (how it is now) or South America or AU, set as default, or to Location (full view) 

When On active sat (Bullseye and Data) an Icon to quickly switch to map for that Sat
and on map and Icon to go back to data.
Thanks

rt-bishop, 2020-02-23:
> Option to centeralise map 3 options should be ample
> as is, Africa, (how it is now) or South America or AU, set as default, or to Location (full view)
> 
> When On active sat (Bullseye and Data) an Icon to quickly switch to map for that Sat
> and on map and Icon to go back to data.
> Thanks

That's an interesting proposition. I'll definitely look into that. Thanks!

rt-bishop, 2020-02-23:
I've just recently released v1.3.0 with search feature, select all/none button and additional sats. By additional sats I mean all currently active sats from celestrak, which is an INSANE amount. Don't forget to update your TLE file via the main menu. I tried to calculate passes for all of them... just be patient, it may take slightly under a minute. Not sure if it's that useful though. I look forward to receiving your feedback. Hope I didn't break anything.

closed 2020-02-23 by rt-bishop

novakmi85, 2020-02-23:
Hi.

I have updated the app in my phone. Looks like everything works fine now.
Great work! Thanks a lot.

Kind regards
Michal