rt-bishop/Look4Sat #5: Feature request: load custom TLE file

ckuethe, 2020-03-14:
This would be useful for spotting satellites for which TLEs aren't available or good enough. Celestrak publishes supplemental TLEs for starlink which may be better than those from JSPOC.

I'd like to be able to load a file of TLEs from a url or sdcard and then from that list select the objects to predict.

Thanks!

rt-bishop, 2020-03-17:
That it definitely a part of my plan. Hopefully I'll get it done within the next few weeks.

ckuethe, 2020-03-17:
Note that some TLE sources, eg. https://tle.info/data/TLE.ZIP, are distributed as zip archives - if automatic decompression is inconvenient I can manually decompress.

rt-bishop, 2020-05-27:
Feature added in recent 389edcb33ae65e41f5b2926c6947cd0bea51d20a commit

closed 2020-05-27 by rt-bishop