rt-bishop/Look4Sat #6: Feature request: Use of compass

Milan12332567, 2020-03-23:
Hi, love the app, but id like the option to enable rotation of "radar" by compass.
Thanks

therman89, 2020-03-29:
Up! A compass would be really handy for handheld satellite tracking.

rt-bishop, 2020-03-30:
Thanks for your request, guys! Probably will add it within the next few weeks, first I need to peek on how other apps implemented this)))

Milan12332567, 2020-04-01:
yeah, i need to track for HRPT so it whuld be great

ryandzink, 2020-05-02:
I'm not sure if this should be its own request or not, but maybe a direction arrow on the radar view showing start and end of each pass? That info is on the pass listing view but not on the radar itself.

rt-bishop, 2020-05-27:
Feature added in recent 80b62b46580735334c8acd2410b7b1831bc8cbbf

closed 2020-05-27 by rt-bishop