rt-bishop/Look4Sat #10: Feature request: Notification of pass start

therman89, 2020-03-29:
A notification of some kind would be helpful, for example at the start of the pass of the satellite or some defined time before that. If enabled, care should be taken about the situation, when ALL satellites are checked!

rt-bishop, 2020-03-30:
Tbh I was thinking several times about the notifications and the way to implement them. Still not sure if I actually want them in the app, for me it was always more of a "real-time" usecase. I'll just leave it open for now, maybe I'll come up with a nice way of doing that.

ckuethe, 2020-03-30:
I know I'd like to get notifications. A few ideas:
* have some filters for which passes will generate notifications... eg. always ping me when the ISS and SpaceX DarkSat are going to be above some elevation.
* have a notification queue, to which selected passes can be added.
* create an event with the alarm clock or calendar apps

When calculating pass times, I'd be interested in when the satellite crosses the horizon, not when it crosses my altitude filter limit.
When configuring notifications, I'd be interested in a configurable amount of lead time (eg. 0-3600s) so I have time to set up my telescopes or radios or whatever.

sbridger, 2020-09-13:
A distinct LED color/flash would be useful when I am waiting for a pass "soon", but don't want to be fiddling with the phone every minute to see how long is left.

Hamalert can beep in morse, which clearly separates its alerts from those of muggle apps.

robsonos, 2020-11-12:
A feature like this would be handy.

shuyuan-liu, 2020-12-06:
For reference, the Heavens-Above app can create calendar events for passes. I think this is a good way of doing it because it means the app doesn't need to run in the background to wait and show the notification.

Auxilus, 2023-12-22:
This would definitely help a lot. We can customize the notifications for certain satellites. This would reduce the number of notifications - especially if someone has a long list of selected satellites. They can choose which satellites they want to be notified for.

closed 2025-10-18 by rt-bishop