rt-bishop/Look4Sat #12: List of passes does not refresh automatically when app is opened

therman89, 2020-03-29:
I need to manually refresh the list after I open the app. I think this should be automatic.

rt-bishop, 2020-03-30:
The issue is fixed in e52e6ec commit. Will be shipped in v1.3.4 by the end of the week.

therman89, 2020-04-24:
I need to reopen this, as the function is buggy. If you use the app, close it, then open it later, the list dies not refresh, instead it shows wrong predictions.
This bug might be connected with the other, that every time you refresh the list of passes, the AOS and LOS times can change by 1-2s every time.

See:
https://photos.app.goo.gl/kBenDUsFYKzPQD4Z8

rt-bishop, 2020-05-27:
> I need to reopen this, as the function is buggy. If you use the app, close it, then open it later, the list dies not refresh, instead it shows wrong predictions.
> This bug might be connected with the [...]

As far as I know the time change by 1-2s is an issue of predict4java lib, but I will look into that issue.

rt-bishop, 2020-09-10:
Fixed in bfefae8

closed 2020-09-10 by rt-bishop