rt-bishop/Look4Sat #13: View refresh rate does not affect tracking view

therman89, 2020-03-29:
The tracking view or single satellite mode data does not refresh, only when I tap back and open again the view.

rt-bishop, 2020-03-30:
I'm not entirely sure why this is happening because I can't reproduce this behaviour on any of the devices I have. The "view refresh rate" setting correctly applies to both the Map view and the single sat Polar view updating the values every 250-10000ms. Would you be more specific on that issue?

therman89, 2020-03-31:
I can show it in a screencast video: https://photos.app.goo.gl/GNcHxUznanmYbmqN7
Watch the azimuth or elevation changing only once after I open the single sat view. When I press back and open it again, it will change only once again.

rt-bishop, 2020-03-31:
I can see the app still sets the initial data straight after the fragment launch, updates it once and then stops. I've never seen this behaviour on any of my devices (testing on ~12 devices I have access to). Is there a chance you can try it on any other phone? Also can you tell me your phone's make and model?

therman89, 2020-03-31:
Strange, I will try on another device, when I can. Mine is a Huawei P9, check the screenshot for details.
[image]

rt-bishop, 2020-04-06:
In the recent update I swapped the Java Executors based view refresh timer for the Kotlin coroutines one. As I've never encountered this bug myself I can't really test whether it's now fixed or not, so I'm just gonna wait for you to confirm if the bug is still there. 

Also, did you try the app on any other device? I had another user with the same issue on Android 6.0, but still I can't find any proper explanation why this keeps happening.

therman89, 2020-04-06:
I can confirm that this isue has been resolved on my device.

closed 2020-04-07 by rt-bishop