rt-bishop/Look4Sat #15: TLE file import filter too restrictive?

ckuethe, 2020-05-27:
Thanks for adding the ability to import custom TLEs in #5. I was just playing with that this morning and noticed a couple of things.

First, zip file import doesn't work from either the network or from local storage, eg. https://tle.info/data/TLE.ZIP. I know I can uncompress the TLE on my phone, but if I'm wishing for things I wish that zip archives containing a single file could be decompressed on the fly, and maybe cached on the sdcard?

Second, only `*.txt` files can be imported. If I had to guess, I'd say it's [MainActivity.kt#L160](https://github.com/rt-bishop/Look4Sat/blob/e38ad2fe8b98ee08945f5ecf0bf87369d3d46579/app/src/main/java/com/rtbishop/look4sat/ui/MainActivity.kt#L160) because android probably doesn't know that `*.tle` is a text file too. I'm no android programmer so I'm not sure how to add a hint that `*.tle` is a reasonable file type to open too.

rt-bishop, 2020-06-15:
@ckuethe thanks for the proposition and sorry for late reply. I'll look into that issue, however I'm not sure yet what's the best way of implementing it and when it's going to be done)

Also AFAIK "type = text/*" doesn't limit the files just to txt, it means any text file in MIME.

ckuethe, 2020-06-15:
No worries, I wrote a little python web server to fetch, decompress, and serve up a custom set of TLEs.

Just reporting the behavior I saw on my phone: `file.txt` was available for import, the same file, renamed to `file.tle` was not. I'm guessing that Android doesn't know that tle is txt too.

SHMAUS-Carter, 2020-07-14:
It may depend on the file manage that you use.

rt-bishop, 2020-09-10:
Fixed in 15d414f

closed 2020-09-10 by rt-bishop