rt-bishop/Look4Sat #16: Constellation view

JonnyTech, 2020-06-05:
Thank you for making this excellent app. A great replacement for the proprietry [SatOrbit](https://play.google.com/store/apps/details?id=sil.satorbit). Can I please request a constellation or planet (or simple sun and moon) view? My device does not have a compass or similar sensor so would help to locate directions. I currently use Look4Sat together with [Planisphere](https://github.com/tengel/AndroidPlanisphere) but would be good to have all of the information on one screen. As Planisphere is open source and on F-Droid then it might be simple to locate the necessary code.

[image]

[image]

Out of interest, are you planning to implement any other features from SatOrbit (sorry if off-topic)?

rt-bishop, 2020-06-15:
Thank you for your kind words and for using the app @JonnyTech! 
I've never used SatOrbit before so didn't use it as a reference. Looked through it and wasn't really impressed, nothing caught my eye. I'm currently working on the OpenStreetMap integration instead of a very simple MapView that we currently have. 
I dind't even know about Planisphere existence. Although it looks quite interesting, I can't really promise you that it will be integrated (probably will) or when it's going to be done. 

By the way, I'm curious what kind of a device you have that doesn't have any sensors at all? Having two apps to do one thing is definitely not cool at all.

JonnyTech, 2020-06-15:
Don't worry about trying to implement what you don't want to, I am just glad that the app exists in the FOSS community, and I appreciate your reply. A correction: parts of SatOrbit are open (https://github.com/Userpack4/Sgp4Silvio) so apologies to the author, but both apps are similar and t

rt-bishop, 2020-06-16:
It's good to receive all this feedback, new ideas etc, it's just they require a lot more time than I currently have. I had a look at Sgp4Silvio lib.. predict4java that I currently use seems superior in a lot of ways, obviously having it's caveats though, which I may fix one day.

I always prefer offline apps myself, so the OSM integration is being done with that in mind. The plan is to use the generic vector map file (mapsforge) that I can bundle with the app, however it will increase the app size for a couple of MB. Alternative way is making cache optional, but I simply don't want to depend on any sort of Internet connection. It's a bit too late to decide to cache the tiles already being in a forest completely offline.

sbridger, 2020-09-16:
@rt-bishop I totally agree with your offline point!