rt-bishop/Look4Sat #17: Ability to group/filter selected satellites

ryandzink, 2020-06-08:
I was wondering if it might be possible to group the selected satellites into different folders. For example, based on what equipment you have with you you might only be able to work FM satellites so you would want to temporarily filter out linear satellites, etc. in the upcoming pass view.

rt-bishop, 2020-06-15:
Good day to you @ryandzink! In order to address this issue I need to rewrite quite a lot of app logic. I'm going to do it anyway, not sure when though, as I develop this app in my free time.

ckuethe, 2020-06-15:
This could totally turn into a couple of very big project:

I'm imagining a database grouped by name, NORAD, or COSPAR numbers that is independent of TLE sources. Then you could, for example select all the starlink satellites. Or all the OSCARs, or ... to display or hide.

I say independent of TLE sources because maybe you want to get the supplemental TLEs for Starlink and the ISS from their operators by way of Celestrak, some cubesat TLEs from the university teams running them, and the rest of the TLEs from NORAD. Each of these providers may not have all (or the very latest) data but taken together you have a pretty good idea of what's up. But just because I'm not currently fetching the starlink TLEs doesn't mean that I want my database to forget which objects are starlink and whether or not I want to see them right now.

On that note what about TLE files with multiple entries for the same object, eg. https://celestrak.com/NORAD/elements/supplemental/iss.txt ? There are multiple TLE records for the ISS with slightly different parameters for future orbits.

rt-bishop, 2020-06-16:
Your ideas sound great, although implementing them means a proper revamp of the predict4java library. Ex TLE import has to be modified with the DB and all these providers in mind, also passes calculation logic requires some work... As for now fetching new TLEs just adds them to the main internal DB, overwriting on conflicts, without even an option to clear the DB. Filtering is simply done by name or CatNum in sat selection menu.

Soooo seems like a lot of work to do, but I'll try to find some time for that.

ryandzink, 2021-04-05:
Hi @rt-bishop it looks like this actually got implemented in this latest release, or at least the ability to filter the satellite selection page makes it much easier to do this. I'd be happy to close this out.

rt-bishop, 2021-04-05:
Hey, @ryandzink! Yeah, finally found some time to implement this. Refactored quite a lot, let me know if anything breaks. Also, just published v2.3.2, fixed doppler shift update that got broken. Will be available after Google Play approves the release.

closed 2021-04-05 by rt-bishop