rt-bishop/Look4Sat #18: Sporadically the pass list shows no passes and an endless "refresh" indicator

ryandzink, 2020-07-26:
I've had this happen a number of times, but can't figure out what circumstances cause it to happen.

When I go to the pass list, I see what looks like an empty pass list and a refresh indicator that seems to go forever. However, the countdown timer at the top is accurate, so this looks like a display bug.

[image]

If I got at the pass list from a specific pass polar view, and I try to go back to it, I get this screen with what looks like default/unformatted strings.

[image]

I haven't found a way to resolve this except by closing and reopening the app.

rt-bishop, 2020-07-28:
Hello @ryandzink! I know of this strange bug, it's quite hard to reproduce though as it just occurs randomly. I think I might've done something wrong with the passes calculation threading. I'll look into it and hopefully fix it by the next release.

rt-bishop, 2020-09-10:
Fixed in f338a05

closed 2020-09-10 by rt-bishop