rt-bishop/Look4Sat #19: [FR] world map zoom

ildar, 2020-07-27:
would be great even without care of datalization

rt-bishop, 2020-07-28:
Hey, thanks for the feature request! I'm still working on the OpenStreetMaps integration for v1.5 which will obviously include this feature. There are some complications but hopefully I'll finish within a couple of weeks.

ildar, 2020-07-29:
That's... shocking )))
What's the purpose for using OSM? I'm sure it'll make apk many times
heavier and there must be a noticeable reason for that.

rt-bishop, 2020-07-29:
Well, not really, the osmdroid library is exceptionally lightweight. Just checked the apk size: 2.6MB vs 2.5MB before. And I haven't even removed the legacy map code and resources yet. I will also limit the max zoom level to 6-7 (4-16MB of tiles cache max) cause we're watching satellites, not the map itself. Tiles are being cached automatically but there will be an option to pre-download them for a full offline experience. 
I wanted some proper maps funcionality for a long time and also it's the first step for the upcoming user interface revamp.

rt-bishop, 2020-09-20:
Feature demo added in recent v1.4.8, more work on it in the future release

closed 2020-09-20 by rt-bishop