rt-bishop/Look4Sat #22: clear satellite data

DC7IA, 2020-08-30:
Great app, finally a decent satellite tracker on Android.

However, there's one thing that bothers me:

I've set the TLE data URL to one that only contains amateur radio satellites, but the list is still huge.

I'd love a way to clear the list and only show the satellites that are included in the current dataset. :)

Something like a button that says "Purge TLE data", so that when I change the URL I only get a list of satellites in the dataset I'm using.

What do you think? :)

rt-bishop, 2020-08-30:
Hey, thanks for the tip. It's a great idea, though I need to clean up some code for that) I'll release it next week.

rt-bishop, 2020-09-10:
The issue is fixed in recent 0d3c9887e8c20bf29ef7719724b35f20b9a4dcda. The TLE data gets cleared and added properly via the sources provided.

closed 2020-09-10 by rt-bishop