rt-bishop/Look4Sat #23: invalid TLE results in "Check your internet connection"

DC7IA, 2020-08-30:
I've tried using [1], but this results in "Check your internet connection".

I then thought this might be because Look4Sat does not know what to do with the header.

So next I tried [2], which worked.

[1] https://www.amsat.org/tle/current/nasa.all
[2] https://www.amsat.org/tle/current/nasabare.txt

rt-bishop, 2020-08-30:
Yeah, I've just checked, the same behaviour is on my devices. It's the header that stops it from updating. TLE update via text files is super sensitive to the content and AFAIK, Celestrak is switching to JSON-type TLE as they are a lot more flexible. So in the future I'll add a proper handling of both. For now I'll just add the nasabare source on top of celestrak source.

rt-bishop, 2020-09-10:
The issue is fixed in recent 0d3c9887e8c20bf29ef7719724b35f20b9a4dcda. The AMSAT TLE source is added by default and overwrites the Celestrak data correctly.

closed 2020-09-10 by rt-bishop