rt-bishop/Look4Sat #24: Satellites currently overhead show up even in future passes

ryandzink, 2020-09-12:
I don't know if this is expected behavior but when you look at future satellite passes of a satellite that is currently overhead, you see the current satellite location on both the current and future passes.

As an example, here is an SO-50 satellite pass showing both the current pass and the next one scheduled, and you can see the current satellite location on the next pass as well.

**Active pass:**
[image]

**Next pass showing current position:**
[image]

I'm not sure if it is intended to be this way, but it did look a bit odd the first couple of times I saw it until I put 2 and 2 together.

rt-bishop, 2020-09-14:
Hey Ryan! To be honest, initially it happened unintentionally, but I really liked it as it helped a lot with testing. So I decided to keep it that way. Also it led me to the idea of making a separate PolarView fragment with all the sats currently overhead, which I'm going to implement some time in the future after the complete UI revamp. Do you think it's an issue, shall I keep it open?)

ryandzink, 2020-09-14:
I don't think it's a big enough issue to be fixed, I think we can close it. That all satellites overhead view does sound interesting!

closed 2020-09-14 by ryandzink