rt-bishop/Look4Sat #25: True north

sbridger, 2020-09-13:
It appears to use magnetic north for the N position, not true north. 

This is currently 23deg different where I live..

I would suggest correcting for Declination so the NSEW grid is showing true.
Explicitly show the magnetic north point on the edge. (a red N on the edge of the circle at the declination) so it is clear that it is true.

I would suggest manual entry of declination as the first step. (and also since it works when online urls become broken)

Beyond that you can download or calculate declination:

//Online Declination references:
//http://www.geomag.bgs.ac.uk/data_service/models_compass/wmm_calc.html
//https://www.ngdc.noaa.gov/geomag/calculators/magcalc.shtml#declination
//https://www.ngdc.noaa.gov/geomag/calculators/help/declinationHelp.html
//sample url//"https://geomag.bgs.ac.uk/web_service/GMModels/wmm/2015v2/?latitude=-37&longitude=174&altitude=0&date=2019-07-02&format=json"

Here is a qml file that was downloading declination:
[MagMap.zip](https://github.com/rt-bishop/Look4Sat/files/5213558/MagMap.zip)

rt-bishop, 2020-09-14:
Hey! Thanks for your request. I'll definitely implement this with the next minor update that should be within the next few weeks. There is a way in Android SDK to calculate true north and declination for the station position, so it's not going to be a problem. I'll also switch to using the software (fusion) sensor rather than the hardware one for more accurate readings. Not sure about the manual declination entry though, but I'll look into it. And I'll check the links you provided.

sbridger, 2020-09-16:
Thanks.
I have a red N on the outer circle edge at the magnetic north position. This makes it clear that the grid is true north, and also makes it apparent if the declination is badly mucked up. 
A potential issue with android api, may be if it gets declination from a remote server, and so doesn't work when internet connection is not available.

closed 2020-11-29 by rt-bishop