rt-bishop/Look4Sat #26: FR: Calculation of current frequency based on Doppler shift

tekk, 2020-09-15:
It would be nice to have current frequency calculated with regards to Doppler shift.
Another useful feature would be if you could "send" the Az, El, Frq etc... via CAT (serial) or UDP (via network) to an ESP8266/ESP32 based board. Simple text/binary values would be sufficient.
Thank you for an amazing project!

sbridger, 2020-09-16:
I was also thinking it would be nice to see a doppler value.

In the case of the ISS, there is every possible permutation of up and downlinks, so you would need to tell it the up and down, or be able to select one of the modes, or be updating the doppler for every frequency in each/all mode paragraphs.

An alternative (or an easy starting place) would be to show generic doppler shift at one or two frequencies , and leave the user to work out the +/- signal frequency.  There could be a doppler line below the alt/dst line e.g
`dF 3.2k@146M  9.6k@438M`

That would help me know that the frequency is not wildly wrong, and it would cover most of the ham sats. It would cover my actual need.

rt-bishop, 2020-09-17:
@tekk Thank you for your kind words! 
Do you need this CAT or UDP info to control some sort of a rotator or external device or just to display? If to control then it's better to use some sort of protocol like hamlib for ex. If just to display then it's better to use some predefined format of a string. Also UDP is pretty straightforward but requires your device to be connected to LAN and to have an IP, so you would need a separate dialog/fragment for that. And by serial do you mean UART/SPI? Over USB? Or simple linux virtual device/file?
The suggestion sounds really great! Although I'm not sure when I can start implementing that...

@tekk @sbridger Guys, thank you very much for your requests! The Doppler shift correction is a must, I agree. I'll definitely add it quite soon, probably with the major UI update that I have in mind and planned for the next few months. The thing is.. I'm getting married in November. There are a lot of things for me to think about at the moment but I'm trying to find some spare time for all that. So be patient and keep suggesting stuff that you would use!)

tekk, 2020-09-17:
Guys, thanks for your thoughts and quick responses.

@rt-bishop The idea here is to have transceiver like Yaesu FT-817/991/857 or something like Icom IC-9700 connected via CAT control cable (which is basically an UART serial protocol) and have the ability to tune to exact frequency based on calculations from the Doppler shift. One way to do this is using an USB-CAT cable. Bluetooth CAT adapters exist too. When you connect such cable via OTG, you get an UART serial device. The CAT protocol for i.e. 817 is really simple, take a look at this Arduino library: https://github.com/8cH9azbsFifZ/arduino_ft817 If we could use hamlib here (I don't know if there is a port/wrapper for Java/Kotlin) we would support even more rigs.
The calculation of the Doppler shift is really not that difficult. Take a look at lines 330-334 in this file from Plan13 library https://github.com/BackupGGCode/qrptracker/blob/master/Arduino/libraries/Plan13/Plan13.cpp and method `getDoppler`.
I already forked Look4Sat in order to have an overview of what would need to be done to implement this calculation, but I'm quite new to both Kotlin and TLE calculations.
The idea about sending UDP packets to an ESP32 was just an idea, as the ESP32 could control both the rig and a rotator.

How awesome would that be? Complete ground station controlled from an Android tablet/phone.

tekk, 2020-09-17:
I've made some progress.
[image]
It doesn't update the values in real-time though.
@rt-bishop Could you please help? I just need to make it refresh every `timeout` ms, like the Polar view.
Please take a look at my last commit in my forked repo here: https://github.com/tekk/Look4Sat/commit/d8ef1c9f400096ad6a3dcade6737ee8e32115f03
The frequency shift calculation was already there, I've just used it.

tekk, 2020-09-18:
Hey guys,
I've made it. At least the Doppler shifted frequency info is shown on the screen and updating every `timeout` ms. I have not checked the calculations but they should be fine.
Here is a [video](https://youtu.be/u6YBRp_7mMo), but it didn't record very well.
@rt-bishop I will made a PR if you don't mind.
This is the first step. Sending of data over a CAT cable (Frq) or to an ESP32 (Frq, Az, El) can be done afterwards.
Thanks for your support!

rt-bishop, 2020-09-18:
@tekk Great job with the PR! I was about to say that predict4java lib has a doppler shift calculation function in it and I wanted to add it at the weekend but you beat me. Thanks for that!
I'll rebase my dev branch on top of your commits as I'm about to release v1.4.8 with some bugfixes, true north auto calculations based on the station location and OpenStreetMaps integration demo. So expect it to be on GooglePlay/F-Droid on Sunday night.

Regarding the phone controlled ground station, that's what I always wanted. Also to be able to use rtl-sdr for weather satellite imaging/telemetry decoding. However, the next few weeks I'd like to spend working on a complete UI revamp with the BottomNavigation, code refactoring, finishing the OSM MapView integration... I'll have to postpone all the other features till I finish. Hope it's not going to take too long. I'm really glad that the app is useful and I'm thankful for all your requests.

tekk, 2020-10-06:
Hey Guys. I've compared results of the Doppler shifted frequencies alongside Ham Radio Deluxe's calculations and they are exactly the same.
Just wanted to let you know that the calculation is correct and you don't have to re-check.
@rt-bishop Any news on Hamlib? Is it doable?

rt-bishop, 2020-10-07:
@tekk Hello again! I guesss it's doable, there is an android port of hamlib in their repo. The first thing - I'm getting married in the beginning of November, so my time and thoughts are consumed by that at the moment. The free time that I manage to find I spend on v2.0 - UI updates, architecture change, code cleanup. Second thing - I've never worked with C-code/NDK in android, so it's going to take me some time to figure things out. I'm planning to start after the wedding and v2.0 release.

tekk, 2020-10-07:
@rt-bishop Ah, so you decided to ruin your life like me 🤣  Nah, just kidding.
OK, so let me know when you'll have some more spare time and maybe I could help with that - I'm a C/C++/C# developer so it shouldn't be a problem. This will be a KILLER feature!

closed 2020-11-29 by rt-bishop