rt-bishop/Look4Sat #27: Countdown-time on satellite/polar view [feature request]

sbridger, 2020-09-16:
I suggest having the countdown timer on the satellite polar view until it reaches 0. 
Since I am awaiting a particular pass, I would rather just leave it displaying that pass, and see the countdown there.

Watching the countdown on the pass-list screen is a bit of a trap as the timer immediately jumps from 0 to the next satellite pass, making it easy to miss the pass start.

Also nice to know which edge it will rise on and or the AOS/LOS info.

BTW, I am adding a number of FRs as I think of them, feel free to ignore them!

rt-bishop, 2020-09-17:
I'm really glad that you post all these suggestions as I'm trying to make this app as helpful and useful as I can. So please keep on!) The PolarView fragment will definitely have it's own timer with the countdown for AOS and then for LOS. Also a little "bell" will be there to toggle the notification/beep of AOS/LOS. I'll try to find some screen space for the additional info that you requested.

sbridger, 2020-09-17:
You can put it slap over the top of the polar , as it is only there until AOS. At AOS it disappears.

A possibility is to put it over/near the AOS end of the track, so it is obvious where the sat will rise. (might be a good idea, or might look a bit messy).

> Also a little "bell" will be there to toggle the notification/beep of AOS/LOS.

I would suggest that if the polar-pass view is selected then bell should be on by default. i.e. if I have selected pass/polar then I am _actually_ waiting for this pass to start.

hayden-t, 2020-10-05:
yes please this would be great

closed 2020-11-29 by rt-bishop