rt-bishop/Look4Sat #32: ISS times are incorrect, out by about 5 minutes!! Location confirmed.

hayden-t, 2020-10-05:
I have verified my location.  And tracked ISS with SSTV transmissions 2 days in a row now. And tracker app is wrong, and ive compared it with other websites to confirm this. Its about 5 minutes too early.

Im located in NSW australia. As an example use coords -31, 152 i put them in app, and then compare with heavens above website, 5 minutes difference. The app is wrong, confirmed by actual radio tracking.

rt-bishop, 2020-10-07:
Hey Hayden! My raspberry pi weather station, my radio observations and I - we all disagree with you. Heavens above and other tracking wibsites are not a reference for me, only Space-Track and Celestrak are. So I compared the app predictions with Gpredict ones, as they both use Celestrak data, and everything is correct. If you said 5 seconds difference, then I get it. But 5 minutes.. means you're doing something wrong.

Freshly updated TLE, same location, the result...

[image]
[image]
[image]
[image]

hayden-t, 2020-10-08:
I just used heavens above as an example, please tell me how to get data you believe ? as Im not sure how to use those websites. please show somewhere i can put in my coords, and get the passes to compare.

hayden-t, 2020-10-08:
please dont just close this ticket without working through it with me

rt-bishop, 2020-10-09:
@hayden-t No worries, if your TLE data is up to date and the location you set in both the app and the website are the same, then you will get the same passes. 
I think I know what the issue is as I've just checked the heavens above website. It considers the "pass start" the time when the satellite reaches the 10 degrees altitude, whether Look4Sat shows the full pass from the moment it rises above the horizon. If you click on the pass to see the details on the heavens above website, you'll see that there is a list with separate events and times for them. Could you compare the data you get and post here what you find? The time difference I get is about 5 seconds.

hayden-t, 2020-10-14:
sorry, so busy, will look into this Look4Sat  website

hayden-t, 2020-10-14:
so it seems my tle data was out of date as doing so has moved iss pass later about gap i reported. which is what i was hoping that id just needed to do something different. so may i suggest some change in the app to make this more clear. as id thought pulling down the list of passes screen for a spinny refresh would have done this. what does that do then ??? also when you do go to tle import from web, it appears to be just a screen to add sources, and you just click ok when they are done. it does not appear to be a screen that actually downloads them, maybe the button should say download/update instead of ok, also maybe a progress meter/dialog ?? as i had no idea it was actually downloading it until i saw a quick flash up one liner that said it had downloaded which is easy to miss or not read in time. also maybe some suggestion on how often this should be updated ? maybe even an option for auto update ? thanks.

ryandzink, 2020-10-26:
For what it's worth, I also had this issue. It took me a few tries of only picking up the satellite when Look4Sat said it was almost at LOS before I compared to other apps and noticed the discrepancy of ~5 minutes and forced an update. I'm not sure if the recent orbit change means the TLE I had gotten a couple of months ago was out of date?

rt-bishop, 2020-10-26:
A couple of months? Well, generally it's advised to update TLEs every week or so, the error of SGP4 is about 1km/day. Of course the orbit change means the TLE becomes out of date. I'm not sure I understand what the suggestions is... do you want the updates to be automated? Or shall I put a message somewhere saying "Weekly updates are recommended"?

ckuethe, 2020-10-26:
"Update your TLEs Weekly" message would probably be easiest for you. 

Auto update after a week, plus allowing immediate refreshes if you add a new source or if you know a source *just* updated the data would be nice

closed 2020-11-29 by rt-bishop