rt-bishop/Look4Sat #36: Footprint incorrectly rendered when sat at high latitudes

shuyuan-liu, 2020-12-01:
In the map view, sometimes when a satellite is at a high latitude, its footprint is not shaded correctly. The area where the footprint should be is _not_ shaded, while all other area above the minimum latitude of the footprint _is_ shaded (kind of like "inverted" if you like). See screenshots.

Correct:
[image]

Incorrect:
[image]

shuyuan-liu, 2020-12-01:
Sometimes it switches between "correct" and "incorrect" on each refresh.

rt-bishop, 2020-12-14:
Hey @shuyuan-liu! Thank you very much for creating this issue. This is a long time known bug (for me)) that I'm having difficulties to fix. I'm trying to fugure out an algorithm of splitting and recreating the array of sat footprint points based on the latitude. I'll try to look into GPredict's code to see if I can find any solutions there. Hopefully with the next update I'll fix this issue and #41 as well.