rt-bishop/Look4Sat #37: v2.0 tabs seem oddly ordered

ryandzink, 2020-12-03:
First, I just want to say I am loving the improvements in v2 of the app.

This seems like a minor usability improvement but the order of the tabs in the app seems inconsistent with what I picture most users would use most often. I assume most users look at the pass list most often, or the world map, but those are not the initial screen that comes up when the app is opened. The satellite list is, but I at least don't modify the satellites I'm tracking very often. 

I'm imagining that that tab is first to encourage users (definitely including myself!) to update the TLE data more often? In my opinion the tabs would make more sense ordered like this (from most accessed to least) so the most used tab is open on app startup:
* Passes
* World Map
* Satellites
* Settings
* Info

rt-bishop, 2020-12-14:
Hey @ryandzink! Thanks for the feedback and for another suggestion! It helps a lot with the development.

Exactly, it was a way to remind the user to update the TLE and to encourage to try different satellites. I received quite a few emails asking to change the order of the tabs and I'll definitely do this with the next update. There is a bit of app logic behind it, so I would need a bit of time to rewrite it. Hopefully within a week or two it will be done. Trying to find some free time for this.

rt-bishop, 2021-03-09:
Fixed in recent 4705ff69158f31c93d4ab198bcd1449c9d95e63f commit

closed 2021-03-09 by rt-bishop

ryandzink, 2021-03-09:
Thanks @rt-bishop !