rt-bishop/Look4Sat #38: Feature Request: Time span filter

BaumGuard, 2020-12-05:
It would be nice to have the option to filter out passes which don't happen in a certain time span. Look4Sat also shows nightly passes, but during the night I don't receive satellites because I'm asleep then.
In addition to that, it would be nice if Look4Sat showed the pass list after starting the app, like in the versions before 2.0.0.

rt-bishop, 2020-12-14:
Hey @BaumGuard! I'll definitely switch back to showing the pass list on start, but the time span filter seems like a very narrow use case. I'll leave this issue open just to get more feedback and to see if anyone else needs it.

BaumGuard, 2020-12-14:
Ok, it was just a suggestion and to me it's a nice-to-have and not a must-have. So I'll not be disappointed, if it isn't added to the app. Look4Sat is a very nice satellite tracking app and what I especially like about it is, that it shows the passes chronologically and not ordered by satellites! So thank you for developing this app!

closed 2021-02-28 by rt-bishop