rt-bishop/Look4Sat #39: Custom TLE Import deletes TLEs downloaded online

rfamp, 2020-12-06:
Thanks for this amazing application, it is great for tracking satellites but there is one thing which seems to be strange:

When importing a custom TLE file from my device, the configured satellites and the TLEs download from the internet are lost.
It seems that importing a custom TLE deletes all other TLEs which is unfortunate if you want to have both at the same time.
I think this wasn't an issue in earlier versions. 
Is that a bug or a feature?

Thanks for your work on this great app.

rt-bishop, 2020-12-14:
Thank you @rfamp very much indeed for your kind words! It feels great to be appreciated) 

Initially it was done to avoid keeping the out of date TLE data. Now the TLEs are stored in the database by their CatNum and are overwritten on update. So you're right, there is no reason to clear the list on every update. I'll fix it with the next update.

judhi, 2021-02-22:
Hello, this app has helped me a lot in tracking satellites. Really appreciate your hard work. 
Recently more satellites were launched and more to come shortly.  Unfortunately Celestrak has wrong TLE, so we need to use manual update. If you would show us the best way to do that in this amazing app, many us in tinygs.com community would be very happy. 😊

rt-bishop, 2021-02-28:
Hey @judhi! So with this recent release I've fixed the sat list clearing on each update. Now you can update the data from Celestrak, Amsat etc and then from any .txt or .tle file in your device memory and all the data will be combined. If there are two or more TLE with the same CatNum, the last one imported gets saved.

closed 2021-02-28 by rt-bishop

bv2dq, 2022-04-19:
@rt-bishop I teach high school students to make Yagi antennas, and it best to use your apk..It works well..Thank you for your contribution.
I have a problem and can't import the TLE file I made, can you help me to find the problem?
[BaoFengTLE 041922.txt](https://github.com/rt-bishop/Look4Sat/files/8510630/BaoFengTLE.041922.txt)

73 de Randson /BV2DQ

rt-bishop, 2022-04-20:
Hey @bv2dq, I'm super glad to hear about this, epic! That was the intention for making the app open-source so more people can get into radio and satcom. The TLE file import is very sensitive to the file formatting. Simply delete the trailing spaces so there are no empty lines in the end of file and the import will succeed. Have just tried it on your file and it works. TLE format is not efficient for several reasons including parsing, that's why the industry is moving towards such formats as csv, json, xml.