rt-bishop/Look4Sat #42: Map view wraps around vertically when zoomed out

shuyuan-liu, 2020-12-16:
If I zoom out in the map view, copies of the world map are displayed below and above the normal one, like they are tiling to fill the empty space above the North pole and below the South pole. This might be intended in Openstreetmap but it does look counterintuitive (for me at least).

[image]

rt-bishop, 2020-12-17:
Out of interest, what's the resolution of your screen? There are settings in the OSM lib to enable the vertical/horizontal infinite map scroll. Both these are disabled in Look4Sat. Curious... it also shouldn't let you zoom out more than the minimal zoom level. I'll do more testing with different screen sizes and I'll fix it in the next release. Thanks for reporting it!

shuyuan-liu, 2020-12-18:
Hi there. My resolution is 1080×2340, 422 DPI. My device is a Xiaomi 9 SE on AOSP Android 11, if that's relevant.

closed 2021-02-28 by rt-bishop