rt-bishop/Look4Sat #44: Valid Locators Are Not Accepted

DC7IA, 2021-01-01:
Hi,

some locators aren't accepted by Look4Sat. It says they are "invalid".

Version: 2.0.1
Locator: JO51BJ

DC7IA, 2021-01-01:
Weirdly enough, it worked a few minutes later for me.

@Konradrunfdunk is the original observer of this bug

DC7IA, 2021-01-01:
Alright, I narrowed it down to being upper-case. While `JO51BJ` does not work, `JO51bj` does.

DC7IA, 2021-01-01:
Look4Sat should tell users how to type the locator or ignore case, imho.

Konradrundfunk, 2021-01-01:
It seams to me that the funktion setPositionFromQth in the class Prefsfragment is stricktly set to thoose values.

rt-bishop, 2021-01-04:
Hey everyone and thanks for reporting this! To be honest I didn't expect the letter case to be such an issue for so many users. I'll definitely fix it with the next update, so the case will be converted internally, accepting those wrongly "invalid" locators. Will also create some unit tests to cover this issue.

closed 2021-02-28 by rt-bishop