rt-bishop/Look4Sat #45: Edit satellites frequencies

ea5wa, 2021-01-07:
It will be nice to have the possibility to edit the uplink/downlink frequencies and also the possibility to enable or not displaying those frequencies.
Sometimes we are only interested in one frequency

rt-bishop, 2021-01-12:
Hey there! Thanks, @ea5wa, for your support and this suggestion! However, to me this looks like a very narrow use case. I can understand when a radio amateur wants to add the freqs missing in the SatNOGs DB, but hiding/showing... I'll leave this issue open for now just to see what other people say and how many actually need it.

ea5wa, 2021-01-12:
Some people are not interested on telemetry or cw beacon. They only want to know the transponder frequencies....

closed 2021-02-28 by rt-bishop