rt-bishop/Look4Sat #48: Antenna tracking from app

ok1cdj, 2021-04-09:
Hello, thank you very much for great app. It will be very intersting to have antenna tracking functionality from the app.
The easy and most compatible way is control `rotctld` over the network. This is very easy to do. From linux commnad line
`echo "\set_pos 135.0 30.0" | nc -w 1 IP 4533`
Is it possible to implement this feature to the app ??

Thank you.

rt-bishop, 2021-04-10:
Hey there! Thank you for your kind words and for this suggestion) Of course it's possible. It is even half-done already. I've written most of the code, just need to integrate this part properly. Hopefully I'll release it within two weeks.

rt-bishop, 2021-04-18:
Feature added in recent e5e5fe7ee771e7affb3f09e54ee13d19f1d65b02

closed 2021-04-18 by rt-bishop