rt-bishop/Look4Sat #50: No GPS reading

SeriousM, 2021-04-15:
When I start the app with location service enabled and try to set the location via GPS, the app toast-notifies me that location services are disabled. 
How can I fix that?

rt-bishop, 2021-04-15:
Look4Sat gets the location passively to avoid unnecessary battery use, it doen't activate sensors to compute the location, it only receives the cached one. When you press the Set position from GPS, the app gets the last known location requested by any other source, apps like Google Maps or the system itself. If the toast says the location is off even when it's on, it means that you've just switched the location service on, but it hasn't got any location updates yet. Try opening google maps or any other apps that actively use your location or even let the phone settle for several minutes and then try to update it in Look4Sat.

SeriousM, 2021-04-15:
That worked flawlessly! I usually have gps turned off because of battery live and privacy reasons.
Maybe you could update the toast message or add some help for other users that encounter the same issues?

rt-bishop, 2021-04-15:
I can understand, I used to do exactly the same. But since I properly set all the permissions for the apps, removed all the bloat apps, switched off the location history and all the other tracking features of a google account, I became kind of reckless and lazy. 
For sure I'll think of a way to distinguish between the switched off location and the lack of cached data. Thanks for bringing this up!

closed 2021-04-17 by rt-bishop