rt-bishop/Look4Sat #51: Satllite dot on radar view becomes two dots as pass goes on

ryandzink, 2021-05-01:
This behavior seems to have appeared in the past one or two updates.

What starts out as a single satellite dot on the radar view slowly splits apart as the pass goes on to show two dots at different points on the arc, with one traveling slightly slower than the other, and one of the two dots appearing "above" the red arc while the other is "below" the arc.

If I back out to the list of passes and then come back in to the radar view, there is once again only one satellite dot on the arc.
[image]

ryandzink, 2021-05-04:
I wanted to update this to say this is an issue with the entire radar view. I usually don't have rotation turned on but when I did I got the following result. This issue seems to be triggered more often when turning off the screen when on the radar view and then turning it back on.
[image]

MondoBurrito, 2021-05-09:
Yes, I get this problem as well. I hope there is a fix soon.

rt-bishop, 2021-05-11:
Hey guys. Thanks for reporting this. I've already fixed it in the dev build. Hopefully I'll ship it in a week or two maximum.

ryandzink, 2021-05-18:
I see this is fixed in the latest release, thanks!

closed 2021-05-18 by ryandzink

MondoBurrito, 2021-05-18:
AWESOME! Thanks so much.