rt-bishop/Look4Sat #52: Feature request: Directional arrow showing the direction of movement of the sat on radar

MondoBurrito, 2021-05-05:
This would be a REALLY nice feature to implement... and likely not too hard.  Apps such as Heavens Above show the direction of movement of a satellite on their radar screen. It is simply a little arrow on the projected satellite path.  Can you please implement the same feature on Look4Sat. It saves us having to go to the map and trying to figure out what direction the sat is coming in. We can simply look at the radar and instantly know the direction it is moving by looking at an arrow. Thanks for the app.

[image]

rt-bishop, 2021-05-11:
Hey, thanks for the RF. I'll definitely add it with the next update in a week or two as this feature seems to be in quite a demand.

pedjas, 2021-05-18:
Similar arrow would be helpful in map map mode too.

MondoBurrito, 2021-05-18:
I see the arrows now! This is such a great feature to implement. Well done. You have one of the best sat tracking apps out there now.

rt-bishop, 2021-05-19:
@pedjas I can add such an arrow, but please could you clarify what's the use case of such an arrow? When you just select a satellite on the map it clearly shows the direction, there is a track for the three future orbits and no track behind it.
@MondoBurrito you're welcome! Thank you very much indeed for your kind words. Trying to make it as useful as possible.

MondoBurrito, 2021-05-19:
> 
> 
> Similar arrow would be helpful in map map mode too.

I think in this case, an arrow is NOT necessary since it is obvious from first glance which direction the satellite is moving. You simply find the sat on the map, and and the patch is clearly marked. (The app does not show any path where the satellite previously was.)

pedjas, 2021-05-19:
> I think in this case, an arrow is NOT necessary since it is obvious from first glance which direction the satellite is moving. You simply find the sat on the map, and and the patch is clearly [...]

That is correct for selected satellite, but map may show several satellites. For those that are not selected user has no clue what direction they are flying until he selects them.

MondoBurrito, 2021-05-19:
> That is correct for selected satellite, but map may show several satellites. For those that are not selected user has no clue what direction they are flying until he selects them.

Gotcha. Good point.

rt-bishop, 2021-05-19:
@pedjas it's not a problem to show the direction arrows, I just simply do not understand how can this information be used? When I look on the map the direction of the sat does not help me in any way. Or maybe I'm missing something, what's the use case for that?

pedjas, 2021-05-23:
When I look at the map I am interested to see which satellites are incoming so I can prepare for them. Actually, I cannot tell other purpose of the map.

closed 2021-05-29 by rt-bishop