rt-bishop/Look4Sat #55: Frequency info feature suggestions

pedjas, 2021-05-18:
I am novice user of this app so forgive me if i did not understand all the options yet. Here are my observations:

Frequencies list for satellites seems to be available only on satellite details screen which is accessible only from Passes screen.

From user point of view, it would be useful to access satellite and especially frequencies info regardless Passes. Thing is Passes list shows each pass so it my be lengthy list requiring scrolling to find right satellite. And it is a bit counterintuitive.

Satellite detailed info could be accessible from map to, for example, by clicking on satellite icon shown on map. 

The other issue is that on satellite details screen,  font size for frequencies list is too big. I do not know if it is some global setting on my phone or else, but it is not nearly as nice as in screenshots on your site. I have fairly big screen but only one frequency mode can be seen. Option to set font size for this may help.

Satellite pass graphical compass is on the other hand unnecessarily big. It can serve purpose if twice, maybe three times smaller, leaving space for textual info which has to be read.

rt-bishop, 2021-05-19:
@pedjas thank you for reporting these issues. The app font scales with the phone screen, but looking at both screenshots from the issues I can clearly see that you've increased the font size globally in Android settings. I think it even says there that changing the default values will affect and may cause problems with the way other applications look.

The map view is mainly used just for the general view of where the satellites are. I think that the transmitters info is only useful if you're trying to catch the upcoming pass and get or transmit the data. There is no use for the transmitter info if you see the satellite flying on the other side of the world. I mainly track about 15 sats myself and find the workflow of the app quite convenient and extremely intuituve for me and for people I know. I may be wrong though, that's why I prefer to receive feedback from the users. 

As for filtering the sat mods, I had a few requests asking for that, but I'm not going to implement that. Looking at the waterfall of the radio signals I always want to know what other signals are in range, and seeing all the transponders gives me that info. I usually select the sats and mods only once and just keep updating them after that. But as I said, my use case can simply be quite narrow and I may be missing other people's use cases. You're always welcome to share the ideas either here in github issues or feature requests or via email.

pedjas, 2021-05-19:
> @pedjas thank you for reporting these issues. The app font scales with the phone screen, but looking at both screenshots from the issues I can clearly see that you've increased the font size [...]

Yes I did. And this is the only application where I have issues. That is why I suggested some graphical optimizations like using abbreviations to consume less space and minimise issues.

> The map view is mainly used just for the general view of where the satellites are. I think that the transmitters info is only useful if you're trying to catch the upcoming pass and get or transmit [...]

Actually, it is useful to access satellite info even if it is not close to you, so you can get info that is needed to set up radio.

> As for filtering the sat mods, I had a few requests asking for that, but I'm not going to implement that. Looking at the waterfall of the radio signals I always want to know what other signals are [...]

That is why configuration options are helpful. One may set application behavior to his use case. I suggested to have filter as an option not as forced feature, so one can choose whether he will use it or not.

closed 2021-05-29 by rt-bishop