rt-bishop/Look4Sat #56: Suggestion for increasing screen readability

pedjas, 2021-05-18:
Here is how Map view looks at my phone. Text is to large and some data is overlapping. Map copyright info is far to large.

[image]

It would help to have option to set font size for text.

Also, shorter generally recognized captions for field names may be used, like Alt instead Altitude, Lon instead of Longitude, Lat instead Latitude, etc...

Option to hide less needed satellite info on top of the map could be helpful. Only satellite ID is really important for everyday use. Other info is nice to see but not necessary.

pedjas, 2021-05-19:
Here is the same font issue in Passes page

[image]

rt-bishop, 2021-05-19:
I'm still pretty sure the reason for the issue is a global increased font setting. Would you please post here you hardware specs or phone model so I can set up a VM with exact same specs to see for myself.

pedjas, 2021-05-23:
My phone is Xiaomi Redmi Note 4 but I guess you can get the same effect on any model, by just using increased font.

closed 2021-05-29 by rt-bishop