rt-bishop/Look4Sat #57: Don't show aiming icon when "Use sensors to rotate PolarView" is disabled

shuyuan-liu, 2021-05-18:
First of all, thanks for adding the aiming feature - very useful for hand tracking satellites. However I think the icon should only be shown when "Use sensors to rotate PolarView" is enabled, because otherwise it is fixed at 0° due north and serves no purpose.

In the screenshot below I have disabled rotating PolarView, causing the aim icon to be fixed at the top of the PolarView:

[image]

rt-bishop, 2021-05-19:
@shuyuan-liu makes perfect sense, thanks for reporting this. I've done like you requested as well as fixed the bugs with the recent 2.5.2 update. I'm glad people find the aiming feature useful. It was a lot easier to implement than I expected, wish I did it earlier.

shuyuan-liu, 2021-05-19:
Thanks @rt-bishop, yes the aiming is quite handy. My phone's accelerometer is off by quite a few degrees though, will need to look for a way to calibrate it.

Reported problem solved, closing the issue

closed 2021-05-19 by shuyuan-liu