rt-bishop/Look4Sat #58: Rename PolarView to Polar View / Polar view

shuyuan-liu, 2021-05-18:
This is a very minor cosmetic issue but I just thought adding a space to the name makes it a bit more grammatical. To me `PolarView` looks like an Android class name haha!

rt-bishop, 2021-05-19:
@shuyuan-liu do you mean to simply start referring to this screen as a Polar View? Where to rename it exactly? I keep renaming the classes quite a lot, cause the app evolves with every iteration. Hopefully one day I'll make up my mind on all the classes names.

shuyuan-liu, 2021-05-19:
@rt-bishop Sorry I forgot to mention where! In the settings page there's an option "Use sensors to rotate PolarView" - I was suggesting adding a space between "Polar" and "View". I don't mind how the classes are named in the code at all! :)

rt-bishop, 2021-05-19:
@shuyuan-liu Ah, yes! No worries) I completely forgot about that little trigger. Of course, makes sense, I'll fix it in the future updates. I need to finally go through all the strings and messages in the app.

shuyuan-liu, 2021-05-19:
Thanks @rt-bishop! On that note, I think it would be a good idea to let people provide translations for the app on something like [Crowdin](https://crowdin.com/) or [POEditor](https://poeditor.com/). I'll open another issue for that.

closed 2021-06-23 by rt-bishop