rt-bishop/Look4Sat #61: Backup TLE data files

Symbian9, 2021-05-23:
Add option to save (backup) actually downloaded TLE data files into custom folder.

FTR, What is default path of downloaded TLE data files on device?

rt-bishop, 2021-05-29:
Hey @Symbian9! Thanks for the request, but would you clarify why you need to back up something that will go completely out of date in a week? TLEs are meant to be updated pretty often to get the correct predictions, otherwise the error quickly adds up. 
Also, satellite data is not stored in a file, it's getting saved to the internal app database on each TLE update.

Symbian9, 2021-05-29:
> Thanks for the request, but would you clarify why you need to back up something that will go completely out of date in a week?

And that is the thing: what about replay old passes using backuped TLE? (*I use Look4Sat for predict spotting of SpaceX Starlink's "trains" passes*).

So, when I saw some sat on the night sky, BUT next day TLE data is outdated or curiosly updated in app, there should be a backup of TLE which could be used for replay.

> Also, satellite data is not stored in a file, it's getting saved to the internal app database on each TLE update.

As I understand, raw TLE text file data converted into database during updating of TLE data sources.

I suggest that there should be an option to backup TLE data by download files into specified folder.

rt-bishop, 2021-05-30:
I'm afraid this is not how TLE works, you can't replay the path simply loading back the file from yesterday. There is no such thing as replay in TLE context. This file simply contains the trajectory data which is then calculated and projected using a complicated algorithm. So there is absolutely no use of extracting and backing up the TLE files. But what I can do is exposing the TIME handle against which all the calculations are made, so the user can control and wind the passes back or forth, effectively replaying whatever pass you want... And that is a really good idea. Noted.

You can replay the pass from yesterday even with today's or tomorrow's TLE. The only difference is - the further your date of calculation is from the date the TLE file was created - the bigger is the accumulated error for your calculation.

Symbian9, 2021-05-30:
> I'm afraid this is not how TLE works, you can't replay the path simply loading back the file from yesterday. There is no such thing as replay in TLE context.

If there is *"Load TLE file from external"* storage (microSD/memory card), so its possible to use some older downloaded TLE data.

So, just download files in some specified (default) folder during TLE updates for backup in raw format (as text files with timestamps in those text file names).

> the TIME handle against which all the calculations are made, so the user can control and wind the passes back or forth, effectively replaying whatever pass you want... And that is a really good idea. Noted.

That is a must have feature.

thkruz, 2024-01-09:
@rt-bishop I know this is old but since it is still open I wanted to clarify:

The TLE contains orbital parameters that are true at a specific epoch (the year/day listed on the first line). The further you attempt to propagate from that time in either direction, the less accurate they are. If this user was trying to replay yesterday, they are correct that the best way to do that would be loading TLEs with the epoch closest to the time they are looking for and then propagating to that time. If they were to use todays and go backwards, it would be as inaccurate as using todays to look at tomorrow - and even less so if the object maneuvered. TLEs don't have any way of accounting for starlink's maneuvering, we just make a new one periodically until it stops moving.

That said, today vs yesterday is somewhat trivial of a difference. As you noted, a week or so is probably a reasonable margin of error for just determining if something was starlink.