rt-bishop/Look4Sat #63: Feature request: Sending of Az/El over Bluetooth serial

tekk, 2021-07-12:
Would it be possible to add support for sending Az/El in preferably in cleartext over Bluetooth serial connection?
We are building a custom rotator and this would be very handy for other projects too.

rt-bishop, 2021-09-20:
Hey @tekk! Finally found some time to work on the app, do you still need this feature? I've looked through the code yesterday and a couple of questions arised. Shall I simply send the cleartext data to a socket of a generic serial board with SPP UUID just like it says here [BluetoothDevice](https://developer.android.com/reference/kotlin/android/bluetooth/BluetoothDevice#createinsecurerfcommsockettoservicerecord)? Shall we use the rotctl/rigctl format? Do you need the frequency reported as well?

tekk, 2021-09-21:
Hey @rt-bishop, hard to tell.
Basically what we would need is to first list all of the available Bluetooth devices that are paired and connected.
Then we have to select the correct one from the list.
Afterwards, it should be just sending of cleartext to the socket, AFAIK.
But I'm not sure, as I'm not very familiar with Android Bluetooth internals.
If this would work, it would be AWESOME!

ajohns1288, 2022-03-08:
I would use this feature too. I think the best way would be for the user to provide a format string and then that would be output over the socket. Alternatively there could be a plugin for each format and the user can select them. It looks like the DataReporter.kt might already have some of this functionality available.

closed 2022-06-03 by rt-bishop

tekk, 2022-06-04:
@rt-bishop You rock, my friend!

Will test this as soon as possible.

Thanks for implementing this feature,
I think many people will benefit from it, since you can simply decode the cleartext Az/El on any MCU imaginable.

Thanks again!
Will let you know when I'll have a rotator ready.

Wish you all the best!
73

Cyfarw9dd, 2024-03-21:
Hi there, I am also testing this feature.  I want to know whether can I use the Bluetooth throught the BLE to connect the board. Thanks a lot!

G0UIL, 2025-07-10:
Hi All
I have built a Satellite Tracker with Bluetooth connectivity it uses a HC-06 module.
I can't get look4sat to connect to it.
Has anyone any idea why this is happening.
Regards Dean

rt-bishop, 2025-07-26:
@G0UIL: I've tested a few ESP32 boards (with HC-06 and a direct BT connection on Espressif) with Look4Sat, the connection and data transfer seem to be working so far. It's hard to diagnose your particular problem without any logs or additional data.

But the work that @tekk is doing sounds like a much better solution. Of course I look forward to seeing that PR! If everything works well then I can make a small release with it, otherwise it may need to wait once I finish the Compose rebuild in develop branch.

rt-bishop, 2025-07-27:
Hey @tekk! I had a look at your fork of Look4Sat to have a rough understanding of the changes that you made. And it may sound a bit rough, but in such a state those changes won't make it into the app.

First, it's an incredibly huge amount of changes. The whole Look4Sat is ~8k lines of code and the suggested improvement is another ~20k lines on top. I can't justify 75% of code increase for a marginal improve of functionality.

Second, it's the quality of those changes. I can see that everything is written in Compose, which is great, however the libraries that were added like material, gson, usb I won't be able to accept. The app uses Material3, so no reason to bring an older material. I'd agree to use kotlinx-serialization to handle jsons, but not the gson library. Regarding the usb, the whole point was to simply send rig/rotctl compliant commands in order to communicate with devices without getting into too many special case handling.

I'm super glad so many people around the world use this app on a daily basis. But handling every special case/device combination is definitely not the goal of Look4Sat. Accepting the code from outside is easy, but it's much harder and more time consuming to maintain that code. Look4Sat being open source and ad-free is my way to say thanks to the open source and radio communities around the world, but I don't see it becoming more than my side project in terms of effort and priorities. I hope you understand.

rt-bishop, 2025-08-17:
Hey @tekk! It's all very complicated to sum up.

Basically I don't want to reinvent the wheel. Personally I've played with Hamlib (rigctl/rotctl) only a few times, I haven't used it for a long time and don't have any current setup. But what I know is that their project hides all the complexity of controlling different radios/rotators and provides a simple interface (one ring to rule them all). Adding any functionality like hamlib already has into Look4Sat seems like an overkill and reinventing the wheel. Initially when I was thinking about extracting the az/el and freq information from the app I thought that this data will simply go via network to the device that will actually perform the controls.

There are many ways to set this up, it can be RaspberryPi with rigctl/rotctl, or Arduino/ESP32 with any custom wiring. And Look4Sat (v3.2.0) already handles this scenarios via Network data output (RaspberryPi way, _including VPNs like Wireguard, SSH tunnels, or similar_ as you mentioned) and BT data output (Arduino/ESP32 way). It just makes sence to retain the rigctl/rotctl compatible commands and if somebody does anything custom, they can simply pipe the data into whatever board/device performs the actual controls. This perfectly adheres to the Pareto principle (80/20 rule) and KISS, reuses the existing frameworks (Linux pipes, Android Intents) and ultimately saves time for everyone. So I'm not sure how much I'd want to expand the app's current capabilities.

HC-06 and direct BT simply work for me with the currently released v3.2.0, network usecases seem to be covered. This HC-12 USB wireless module sounds like an absolute fun to play with. Although making a rotator system based on that seems too custom/low level than needed. I'd definitely play with that, but I wouldn't be supporting or including such a thing into the project.

There could be a separate Android app that leverages Hamlib and gives a way to control everything from a mobile device. This app could expose an interface (via ForegroundService and Intents) to other apps and it would make live easier for many devs out there.

At least for now this is what I think. I may be getting into HRPT and other L/S band satellites in a foreseeable future and will need a rotator. Most of my SDRs are 1500 km away from me and I connect daily to the RasPi based station over there. If I had the rotator already, I'd simply install rigctl/rotctl on that Raspberry and used the existing Look4Sat's network data output capabilities.

bi9ajk-design, 2025-09-19:
Hello everyone, my LooK4Sat software. I want to connect my HC-05 module via Bluetooth, and then the HC-05 will transmit data through USART to the STM32 and control the antenna rotator. However, my LooK4Sat seems unable to connect to the HC-05 module.

commit 278c0617 (2022-06-03): Minor fixes to strings and ui, crash fixed #63