rt-bishop/Look4Sat #68: Feature request: ability to toggle from pass view to map view for active satellite

ryandzink, 2021-09-19:
I think it would be very useful to be able to switch from the radar view of a current pass to the world map view showing the current footprint of the satellite. Right now going to the map view to view the current pass requires scrolling through the list of satellites to find the satellite overhead, and I can't tell what order they are in when cycling through them in the map view with the left to right arrow buttons. Maybe a button on the radar view that toggles over to the map view with the current satellite selected?

rt-bishop, 2021-09-20:
Of course, this feature was actually planned for quite a while now. It's taking me quite a lot of time to clean up the codebase, properly decouple and refactor everything. I'm trying to get the app ready for new features, that's why it's delaying stuff. 
It will be done in a way you suggested with a button on the radar view that takes directly to the map.

Also, at some stage I'll add horizontal layouts for landscape devices, custom pass calculation time, current transmitter frequency reporting via network socket to control the GQRX and other hamlib compatible apps, and other minor things. Stay tuned!

ryandzink, 2021-10-30:
Thanks @rt-bishop ! Very useful feature added and a nice surprise on the latest app update!

rt-bishop, 2021-10-30:
@ryandzink ! You're very welcome... I'm planning a slight change of the UI in the next update. Hopefully it'll simplify the app workflow for some users and make it more straightforward.