rt-bishop/Look4Sat #75: Can add function output satellite angle by wifi? and how to change the world map API to another?

EeeLo, 2021-11-01:


rt-bishop, 2021-11-04:
Hey @EeeLo! Thanks for using the app..)

You can already send this data via wifi or any other LAN, just go to settings -> enable rotator control -> set rotator IP -> set rotator port, and then when you observe a pass satellite's azimuth and elevation are going to be transmitted in rigctl format. Basically you can control any rigctl compatible rotator or software. 

Regarding the map API... I want to keep the app completely open source without the need of any API KEYS. So there is no support of GMaps (you'd need to have Google Play Services), MapBox or any other provider apart from OpenStreetMap.

EeeLo, 2021-11-06:
thanks so much ,I using it and it can send angles to server, very good software,thanks very much

rt-bishop, 2021-11-06:
You're welcome @EeeLo! I'm very glad you like the app..)

closed 2021-11-06 by rt-bishop