rt-bishop/Look4Sat #77: Wrong passes with Id 40069 with 2.5.9

dg9bja, 2021-12-18:
Hi,
I am using that great tool to check NOAA Apt (15,18,19) passes and also Meteor M2 (Id 40069).
Since one of last updates my M2 passes are all wrong. I double check it with other tool's. At the moment I don't know what to do.

Another issue I get after every update: there are always other satellites in my list like QO100 or FO29 and some other. Are this some defaults that come with every update?

Ludger

rt-bishop, 2021-12-18:
Hey, @dg9bja! Thanks for reporting this issue. I'd recommend to do a clean app install. I also noticed this behaviour, tried to look into the code but couldn't find anything that would lead to this. I rolled back to 2.5.7 and obviously the error was gone. But after I reinstalled 2.5.9 from Google Play the data reported was correct. There may be some sort of data collision but I can't reproduce it anymore. I keep cleaning up the codebase so hopefully there will be no such things in the future releases.

Yes, there is a default satellite list and yes, I keep receiving complaints about that. I'll definitely fix this with the next update.

dg9bja, 2021-12-19:
Thanks, I reinstalled it and all was fine.

closed 2022-01-09 by rt-bishop