rt-bishop/Look4Sat #78: Feature request: own short comment to each tracked satellite

egislev, 2021-12-30:
How about possibility to add short user-definable 1-2 lines comment for each tracked satellite for PL tone freq., last worked station, active transponder etc ... ?

rt-bishop, 2022-01-02:
Hey, @egislev! I guess I can implement this. Though it's not going to be a priority =) Tell me more, which way you would like it to be done? Custom message for the satellite or the satellite's transmitter? To be seen on the passes tab or on the radar tab?

egislev, 2022-02-13:
Hi !
I think to insert one additional line at end off each sat info at main sat list window. Then long press on these lines enters into this line edit mode. Then entered in sat tracking window this line will be fine to see under compass/sat name. Also this feature can be set on/off at setting menu.