rt-bishop/Look4Sat #79: Ability to remove TLE entries imported from a file

ryandzink, 2022-01-02:
I recently used the file import method for the first time to add a satellite (XW-3) before it became available in the Celestrak feed. Now that it is available there I don't see a way to remove the manually imported satellite. It's not a big deal since I have already removed the manually imported TLE from my satellite list, I'm just curious if that delete functionality exists and I am just missing it.
[image]

rt-bishop, 2022-01-03:
Hey @ryandzink, thanks for the question! There is no such function, the app simply replaces the last imported data over the old one using the satellite CatNum as a unique key. TBH I don't see a reason to implement a single satellite deletion from db, some users will delete several satellites and then will see them again after the TLE update. Although I can simply add a button to clear the DB if it's needed. There should be a balance between the ease of use + the data update + consistency on one side and the fidelity of data control in the DB on the other. Tell me what you think.

P.S. Hope you're having a nice festive season)

ryandzink, 2022-01-04:
Hey Arty, thanks for the response!

To be honest I just resolved this by clearing the app data in the Android settings and then reconfigured the satellites I wanted to track manually. I have to imagine most users aren't heavy users of the imported TLE file function (but maybe I'm wrong?) so I'm not sure it would be worth adding any functionality to do this inside the app...

closed 2022-03-02 by rt-bishop