rt-bishop/Look4Sat #86: Feature request: Space-Track.org Auth

ctand, 2022-03-13:
It would be great to be able to update TLE using object lists made in Space-Track.org. They have developer info on how they'd best like their databases queried and methods of authentication. I don't think this would violate any of their terms.

rt-bishop, 2022-03-16:
Hey @ctand, thanks for the suggestion. Could you please clarify your request? As far as I know Celestrak provides the same data as does Space-Track. The only difference is that the data gets updated on ST more frequently. Having login on the website is another show-stopper for me, I prefer to not have any login/profile features within this app and to not store any personal data.  

If you siply want to get the categories feature from the ST api, that's a different question and can be implemented without the ST, but with the generic Celestrak api. I can potentially do that.

Devs already have a possibility to add objects on request. On their request to the data provider, such as Space-Track or Celestrak that copies later from ST. Look4Sat only downloads, stores and displays the data coming from the data provider. I don't have any plans on providing a two-way communication with the data providers. And you can always create a TLE/TXT file with the object's orbital data and then import it to Look4Sat via the "import from file" functionality in Settings menu.

ctand, 2022-03-23:
@rt-bishop Hey, sorry it took me a bit to respond. CelesTrak does not provide the same level of data or organization but it does look like what I want is available from CelesTrak in a roundabout way. Following through on my request would require local credential storage and I understand the hesitancy. 

One can make custom groups of tracked objects in Space-Track that can be called via their API but it requires credentialed access. I'll let you close this ticket as you see fit. I'm sure it wont be the last time something like this comes up though given the ~~deficiencies~~ quirks of other TLE sourcing platforms.

### The Problem/My Solution:
I can't seem to find a group of sats in the 5#### NORAD CAT ID range (within the range for sats officially not yet designated) in any premade/published CelesTrak dataset. The sats are however in the CelesTrak database. If you manage to track down the international designation for a sat's launch you can add all the objects from that launch to the app in a single query. Alternatively, you can add each sat individually as single API queries using their 5#### NORAD CAT IDs. I opted to just add the intl designation since it included all 8 sats I was looking for in a single query.

### Background:
Space-Track is much more helpful than Celestrak for finding new AMSAT sat NORAD CAT IDs. I found this to be especially true with the Tevel sats because you can easily sort by launch date to find the intl designation for the launch and use those as a reference point check believed received telemetry from SatNOGS. Even the CAT ID put out by AMSAT recently looks to be incorrect or only partially correct for Tevel. All eight objects are of course being tracked individually and two are separated enough from the other six to definitely be unreachable if you try to catch a pass using the AMSAT suggested TLE.

### Related Potentially Problematic Tangent:
SatNOGS currently uses a temporary 9#### NORAD CAT ID range for some reason for sats officially not yet designated even though the 5#### serves the same purpose for NORAD as far as I understand it. This 9#### NORAD CAT ID range is especially not helpful for querying other databases because it doesn't appear to relate to anything other than what is being ingested by SatNOGS yet they refer to it as the NORAD CAT ID. The proper ID can only be found on SatNOGS by filtering by its 5#### ID after clicking "All Satellites" which is for some reason hidden under the unchecked by default "Followed ID" column which is much more confusing than just using an internal identifier to keep track of objects with changing names/IDs. 🤦‍♂️

My SatNOGS frustration is related to this issue below. This feature might be a bug in your app at some point in the future. 🤡
https://gitlab.com/librespacefoundation/satnogs/satnogs-db/-/issues/245

ckuethe, 2022-03-23:
I, too, would like to see spacetrack query support. Perhaps I misunderstand the problem, but [AccountManager](https://developer.android.com/reference/android/accounts/AccountManager) seems like it could be useful storing ST creds.

Since ST uses a REST API, and they have a query builder tool on their website to make it easy to query only what you need, you wouldn't need to store other profile information. Everything is encapsulated in the URL.

One hacky workaround is a local proxy server that can read arbitrary data sources to build a custom TLE feed. Yeah, not really scalable, but it serves my needs.