rt-bishop/Look4Sat #94: Satellite pass radar view showing odd artifacts since ~v3

ryandzink, 2022-06-22:
I am not sure if this is an isolated issue but I am often seeing pass arcs look very strange, starting roughly since v3 of the app was released a few months ago. This never seems to happen when just opening the radar view from the pass list, only when already on the radar view and returning to it from another app or when unlocking the phone when it was previously on this radar view.

I am using a Pixel 5a on Android 12. Here are several examples of what I am seeing, it happens ~5% of the time, but enough to be noticeable.

[image]
[image]
[image]

ryandzink, 2022-07-09:
This is very interesting. I can reproduce it within a few seconds on my phone very reliably. Let me try clearing my cache, etc. and seeing if any of that has any effect.

ryandzink, 2022-07-09:
OK, that didn't have any effect. I cleared cache and it is still happening. For reference I'm using a Pixel 5a and Android 12.

MrARM, 2022-07-15:
Can also confirm this bug happens on my Samsung Fold 3 running android 12, it seems to happen while I'm using the pointer for extended periods of time.

ryandzink, 2022-07-16:
I'm not sure if this most recent screenshot might point to a root cause, it seems like the pass is trying to draw on a circle that is offset from the radar circle...

[image]

dl3jop, 2023-09-09:
I can reproduce this issue reliably by choosing the pass view and putting the app in the background, most of the time when I reopen the app from background, the view shows these artifacts.