rt-bishop/Look4Sat #98: Radar view doesn't initialize

TripleWhy, 2022-08-20:
When I click on any sattelite, I get what I assume is an uninitialized radar view. I can't do anything either, except for pressing the android back button.

[image]

rt-bishop, 2022-08-20:
Hey @TripleWhy! Thanks for reporting this. I wonder how did you get there? Please provide some more details as I can't reliably fix something I can't recreate myself on any of my devices. Are you on the latest version, 3.1.1? How do you reproduce the bug? Does it happen occasionally or all the time? Did you try cold start of the app (swiping the window from the Android's overview)? Did you try clearing the satellite database, updating and refreshing the calculation again?

TripleWhy, 2022-08-20:
Sorry for the brief description, I was using said phone to report this. The program version is 3.1.1, installed from F-Droid, the phone is on Android 12. Apart from that, I don't really have much more information. I set my position to GPS, the data provider to online, selected all satellites and clicked one from the list. That happened with every satellite, and not just briefly.

However, now, some time after the report, it now suddenly works, and I can't reproduce it anymore. I tried deleting user data and cache from the android settings, but it still works. 🤷

rt-bishop, 2022-08-20:
Hmm, I'm super glad that it's resolved 😁 but we can leave this issue open for now, just in case the bug comes back. I'm reviewing the code at the moment but can't find anything suspicious that may led to such a behaviour. Anyway, the most important thing is that it works now))

closed 2022-10-14 by rt-bishop