rt-bishop/Look4Sat #99: Using sensors for rotating radar view doesn't work on some devices

EugenGez, 2022-09-02:
Hi together.

The following issue applies for Samsung Galaxy A3 2016 / SM-A310F / Android 7.0

When "Use sensors to rotate  radar view" is active, then the red cross is only showed at azimuth: 0°, elevation: 0°. I successfully tested the sensors with other apps like HamGPS, so they are working properly.

On some other smartphones, including Samsung Galaxy J7 / SM-J730F / Android 9, this feature works correctly.

Thanks

rt-bishop, 2022-09-02:
Hey @EugenGez, thanks for reporting this issue! 

Look4Sat is currently using the google recommended sensor-fusion way of getting the data from the phone's sensors. The old legacy way is using the accelerometer and gyroscope which may be missing on certain phone models. Also AFAIK Samsung tends to change the default behaviour of certain system components which may lead to this issue when sensor-fusion works on some phones and does not on the others.

I'm not sure that switching back to acc/gyro is a good idea because it can lead to the same behaviour on the other users phones, let's say working on your devices but stopping to work on theirs. I can definitely experiment with this at some stage, potentially releasing it and then reverting if I start receiving complains. 
The other problem is that I don't have any Samsung devices that I can test on, so even this fix may not work for you Samsung if the company altered the SensorManager behaviour or the sensor name.

EugenGez, 2022-09-02:
Hi @rt-bishop,

Maybe it would be possible to implement both ways to additionally support older devices as well? The decision about how to access the sensors could then be made by the application itself at start time or even by the user in the settings.

I'm a developer too and I know, it could be much work, since You most probably would need to implement an abstraction layer for that two access ways, which also could cause new bugs etc. But hey, on the other side, I think many HAMs are not in that age anymore, that they always buy the newest smartphone. 🤣

I think it would be helpful for many users.

Thank You