rt-bishop/Look4Sat #100: Where is the Earth 3D view?

Jordi100, 2022-10-05:
Hello:
Thank you for this amazing app.
I have downloaded the app from Google Play and it has an option to see the Earth in 3D View.
But after download the code, I can not find this 3D view.
Please, where is it?
Regards,

Jorge

rt-bishop, 2022-10-09:
Hey @Jordi100! Thanks for the issue =)

There was never an option to see the Earth in 3D in this app. Also, I don't plan to implement such an option. The app is using [OSMDroid](https://github.com/osmdroid/osmdroid) as a library to display the Earth in 2D, but 3D never existed. I'm not sure where did you get this info from as it's not mentioned on GooglePlay or F-Droid repository.

Jordi100, 2022-10-14:
Hello rt-bishop,
Thank you for your answer.
I downloaded another app thinking it was yours. Sorry for the confusion.
Best regards,

Jordi

closed 2022-10-14 by rt-bishop