rt-bishop/Look4Sat #104: Bluetooth dont work on android 11 with playstore version but....

F4HTB, 2023-02-18:
Hello Dear!

I actualy working on a portable rotor esp32 controled.
I try to communicate by bluetooth.

I tested several ways and several codes on esp32 to simply make a mirror between the serial port and the bluetooth serial port.

I saw that in your code login with uuid 00001101-0000-1000-8000-00805f9b34fb, that's good.

https://github.com/rt-bishop/Look4Sat/blob/devel/app/src/main/java/com/rtbishop/look4sat/presentation/radarScreen/BTReporter.kt

I used the application androidBluetooth Finder, Scanner Pair from Zoltan Pallagi to check the id and uuid of the services provided by the esp32 base codes and they are present.

I used an application called "Serial Bluetooth Terminal 1.4" on android to test my code and I got the return via the serial port.

At this point, I can't get any feedback from look4sat ...

So I installed android studio and the sdk for android 11 then I git your repo and compile.
On the compiled version with all updates, it worked on my android 11 phone.

I uninstalled and reinstalled from play store. It does not work.

I am not a specialist in android applications... It is possible that you have to recompile for each version of android?
If so, can you update your application to android11?

Thanks in advance.
73! F4HTB

F4HTB, 2023-02-18:
If i can made some request here, if it's not so complicated:
-Add something to see if remote controle over network is connected.
Like change the color of "Enable remote control" to green when connected based on network event?
-Add same to "Enable output" on Bluetooth.
-Add  fun disconnectBTDevice(deviceId: String) in BTReporter.kt to properly disconnect when you switch button
This can maybe makes the app look more professional?

Of course, thank you very much for the work already done.

rt-bishop, 2023-02-24:
Hello @F4HTB! Sorry for the late reply, was a bit stuck with my day job)

I can see you did a whole proper investigation, well done! Actually the BT piece of functionality was written by ajohns1288 and merged [here](https://github.com/rt-bishop/Look4Sat/pull/87). 
I never really looked into it properly as I didn't have a usecase and also devices to test it on, but from his notes I assume it's to do with the permissions and the way they are handled by different Android versions. Google introduces more restrictions with each version. 

Funny that you said that if you compile the app from source, then it works properly. I would try to play with the permissions via Android settings. Currently the app is built to target the latest Android 13, so newer version than 11. But it supports everything down to Android 5. For obvious reasons I can't target an older version, but I will look into the issue this weekend.
I don't promise a quick fix though =)

Also, the ideas you mentioned are really good. I need to set aside some free time to review the app's code, which hasn't happened for a while. Anyway, I hope I'll fix it and it'll be a useful tool in your interesting journey. Rotators are so bloody expensive...

F4HTB, 2023-02-25:
Hello!

Like you, I had also thought about the authorizations to use bluetooth. Maybe if the app is uploaded via the sdk, it overrides all permissions requests?
As I currently have my compiled version is functional, I'm obviously not waiting, so no rush.

As soon as I finish my version of the handheld rotor, I'll pat it and let you know. I'm thinking of a budget of less than 100USD.
It is based among other things on these works https://www.sarcnet.org/sarctrac.html

I use ideas of mechanical implementations and I redid a code since I do not use the same modules and do things. Among other things on the fact that they use a gyroscope module and compass that they posisione at the end of the antenna. 

I only use it for elevation because I positioned the sensor close to the main box and the gravitational field is not affected by the magnetic effects of the motors. 
For the azimuth, I look for the north with the compass sensor with an elevation at zero I manage to calibrate it, then I switch to a position detection with a hall effect sensor because as soon as I introduce an elevation on the magnetic  sensor, the motor fields are too influential and it would be necessary to permanently compensate and create a calibration matrix.

In short, suddenly your software is very useful since many people including myself have a smartphone. It was easier in practice to connect your phone to it in bluetooth instead of take out a computer. And above all it works very well :)

73! F4HTB

BH2VSQ, 2023-08-05:
Hi! This is BH2VSQ, I'm trying to use the Bluetooth features of Look4sat. How can I setup my devices to receive the data from Look4sat?

F4HTB, 2024-08-08:
Hello Dear All, 

I just took my project out of storage. I still had the app running on version 3.1.2, but it have a scheduling issue. 
I uninstalled the version compiled with Android Studio and tested the latest version 3.1.4 directly from the Play Store. 
This time it works!

Many Thanks for your app!

closed 2024-08-08 by F4HTB