rt-bishop/Look4Sat #107: [Satellite Request] GOES 16 (GOES-R)

jtfletchbot, 2023-05-19:
I can't locate GOES 16 (GOES-R) in your app. I did however find GOES 17 (GOES-S).

I tried enabling all modulation types, multiple TLE database updates, and checking multiple categories (Geostationary, Weather, and All).

Am I looking in the wrong area, or does this app not support GOES 16?

Thank you for your time.

rt-bishop, 2023-05-21:
Hey @jtfletchbot! Thanks for reporting the issue. I think it's connected to another newly created issue regarding the data update. https://github.com/rt-bishop/Look4Sat/issues/108 Apparently Celestrak changed the data URLs for TLE files and the app can't currently download the fresh satellite data. 

As a workaround you can manually download the TLE file using the URL like https://celestrak.org/NORAD/elements/gp.php?GROUP=active&FORMAT=tle, save the file to the phone and then within the app try to udate the data offline pointing to that file. You should then see GOES 16, although the satellite category selection will probably not work. Simply search for it by name or NORAD CatID.

We can continue the conversation in https://github.com/rt-bishop/Look4Sat/issues/108 as there seems to be more people involved.

rt-bishop, 2023-05-21:
Glad the update issue is resolved. I guess, we can continue the topic of GOES 16 here. I've just downloaded the updated version and did a clean DB update and here's what I've got. It's searchable by the NORAD ID as well... not sure what to say.
[image]

jtfletchbot, 2023-05-21:
I tried clearing the database in the app and re-updating and still couldn't
see it. Or the other GOES sats except for GOES 17.

I just forced stopped and cleared all its storage. Now they seem to appear,
as well as other satellites I couldn't see before.

Which, upon further testing, is because the app can't tell that GOES 16 has
any transceivers, and I had transceiver filters on. I don't know why I
didn't check that earlier, but I guess I got there in the end. (I did try
selecting all filters at one point, but I didn't try selecting zero
filters. Lesson learned).

GOES 16 isn't the only one that doesn't have any transceivers listed. I
haven't made an exhaustive list, but there's quite a few it seems.

However, the general categories seem to work (Geostationary, Weather, etc).

Technology sure is funky sometimes.

rt-bishop, 2023-05-22:
Yeah, the filtering can be quite tricky sometimes. I think in the future version I'll move the transponders selection to the passes filter dialog, so there is less confusion and satellites in the main selection list are visible at all times.

I wish we received the transponders data for all the satellites our there, but actually the app combines pieces of info from 4-5 different data sources. Some of them provide only TLEs, the others only transponders info. SATNOGs in general is a super cool project, but is community driven and some info is missing. I think it's possible to register with them and add the missing info.

I wish we could receive GOES here in London.

closed 2023-06-03 by rt-bishop